clockworkpi/PicoCalc #1: MMbasic source code

robert-rozee, 2025-03-14:
hi,
    i see you are distributing a .uf2 file containing what appears to be a modified version of MMbasic. will you be publishing the source code to this, or at the least the portions of the source that you have modified?

cheers,
rob   :-)

cuu, 2025-03-15:
Sure thing
Every program will be open source

Just pushing the patches one by one 

as now , you can find the MMBasic,which is PicoMite.patch, source code under Code/

titimoby, 2025-04-01:
You could create a dedicated branch so everyone can help and still not generate confusion for newcomers. 👍

madcock, 2025-04-04:
Totally unofficial, but I forked my own version of the official PicoMite/MMBasic sources and manually applied the changes made in the ClockworkPi patches. Can't be 100% I got everything correct, but it builds and boots for both Pico and Pico2. And it's using the most up to date version of the PicoMite/MMBasic sources (and I hope to keep it in sync at least for the near future).

One bug has already been identified and fixed, as pointed out by people on the forums (input was incorrectly handling end of lines).

https://github.com/madcock/PicoMiteAllVersions