clockworkpi/PicoCalc #6: Specifications for the 18650 battery?

WillMarchant, 2025-03-22:
Is there information on 18650 battery specifications? Battery physical size? With or without protection circuit? Battery capacity? Battery chemistry?  Maybe there's already a recommended battery?  Thanks in advance for any info.

HoofedEar, 2025-03-29:
If it helps, I ended up ordering two 18650 batteries with these specifications:

> Protected Button Top
> Nominal Capacity: 2600 mAh
> Nominal Voltage: 3.7V
> Discharge End Voltage: 2.75V
> Max. Continuous Discharging Current: 2C
> Dimensions:
> 68.90mm (L) x 18.50 mm (D)
> Protection Circuit (Over charge, over discharge, over current, and short-circuit)

I put them in today and it seems to power on just fine. But the more I read about 18650 batteries the more nervous I am about them. So it would help if ClockworkPi could release more details about batteries like:

- Recommended specs
- Is it safe to have both battery slots filled?
- Is there an LED indicator light that lets us know if the device is done charging?
etc etc

WillMarchant, 2025-03-29:
Thanks for that info!  I did get a reply to my email request to technical support:

We carefully designed the battery holder to ensure compatibility with various 18650 battery sizes: Ø18 ± 0.5mm diameter, 65-69mm length. Please check the polarity when installing the 18650 battery. Incorrect installation may damage the device and the battery, or even cause an accident. We also recommend purchasing* the battery from a reliable distributor.

altgreen64, 2025-04-01:
And what about these devices, picocal will be able to handle them?
https://milkv.io/duo
Luckfox Lyra ??

potoo-cz, 2025-04-01:
> @altgreen64  Luckfox Lyra

https://forum.clockworkpi.com/t/luckfox-lyra-on-picocalc/16280