clockworkpi/PicoCalc #7: PicoCalc_PicoMite_v1.0.uf2 在 pico2w 上不起作用

2yanyi, 2025-03-26:
看到官网介绍可以使用 pico2w，我的设别到手后就直接安装了 pico2w。

adwuard, 2025-03-26:
芯片内部的核心代码和工具链不一样。cmake 要改一下增加 PICO_BOARD 的定义。
demo 代码要重新编译的
https://forums.raspberrypi.com/viewtopic.php?t=376508

2yanyi, 2025-03-26:
> 芯片内部的核心代码和工具链不一样。cmake 要改一下增加 PICO_BOARD 的定义。 demo 代码要重新编译的 https://forums.raspberrypi.com/viewtopic.php?t=376508

感谢回复！我修改了 CMakeLists.txt 中的 set(PICO_BOARD waveshare_rp2350_one) 如此配置，并成功 build 得到了 build/PicoMite.uf2

```
# make -j8
[  3%] Built target bs2_default
[  3%] Built target slower_boot2
[  8%] Built target bs2_default_library
[  8%] Built target slower_boot2_library
[100%] Built target PicoMite
```

但是这任不管用，我是否是遗留了某个步骤？设备开机后无任何显示

adwuard, 2025-03-26:
`waveshare_rp2350_one`  的 Board 接口定义和标准 Rpi Pico 2 是不一样的。试试   
```cmake
set(PICO_BOARD pico2)
```

adwuard, 2025-03-26:
我帮你查了，PicoMate 版本 `v6.0.0+` 才兼容了 Pico2。Clockworks PicoCalc的代码是 PicoMate `v5.08`。
大概率会有问题。自己要uprev 一下PicoMate代码解决 `.patch` 冲突。

adwuard, 2025-03-26:
@cuu Any plans for Pico2/Pico2W PicoMite support?
The current current PicoMite V5.08 [source code](https://github.com/cuu/PicoMite.git) seems to be too old for Pico2 builds.

cuu, 2025-03-26:
> [@cuu](https://github.com/cuu) Any plans for Pico2/Pico2W PicoMite support? The current current PicoMite V5.08 [source code](https://github.com/cuu/PicoMite.git) seems to be too old for Pico2 builds.

plan是有的，目前还在忙别的project事情，等有空了我会看一下

adwuard, 2025-03-26:
@cuu @2yanyi 我这边把代码升级了，也解决Merge冲突了。也成功在 Pico2 上面跑起来了。

分支：https://github.com/adwuard/PicoMite/tree/edward/picocalc/picomite-v6.00.0

多测测，没问题了我就正式提交PR。

[image]

altgreen64, 2025-03-26:
Has anyone tried programming RISC-V on picocal
https://www.raspberrypi.com/news/risc-v-on-raspberry-pi-pico-2/

cuu, 2025-03-26:
> [@cuu](https://github.com/cuu) [@2yanyi](https://github.com/2yanyi) 我这边把代码升级了，也解决Merge冲突了。也成功在 Pico2 上面跑起来了。
> 
> 我的分支：https://github.com/adwuard/PicoMite/tree/edward/picocalc/picomite-v6.00.1
> 
> 多测测，[已创建提交PR](https://github.com/cuu/PicoMite/pull/1) 。
> 
> [image]

厉害

有个历史原因需要说明 

当初我fork PicoMite的时候上游 的github repo 还在

但是等到PicoCalc要release的时候发现 上游 github repo消失了，原作者直接删库，取而代之的是一个新的库

https://github.com/UKTailwind/PicoMiteAllVersions.git

所以这里的 PicoMite更像是一个为了 配合 出厂sdcard 版本而做的备份

我会稍候编译uf2 ，测试一段时间没问题的话我会 把这个  新的 PicoMite 6.0部分的代码 merge 到 PicoMiteAllVersions

cuu, 2025-03-26:
> Has anyone tried programming RISC-V on picocal
> https://www.raspberrypi.com/news/risc-v-on-raspberry-pi-pico-2/
> 
> śr., 26 mar 2025 o 09:49 Hsuan Han Lai ***@***.***>
> napisał(a):
> […](#)

No, Not yet

cuu, 2025-03-26:
https://forum.clockworkpi.com/t/anybody-have-interesting-to-compile-firmware-for-rp2350/16237/13

adwuard, 2025-03-26:
> 但是等到PicoCalc要release的时候发现 上游 github repo消失了，原作者直接删库，取而代之的是一个新的库

我这个分支同步的代码，也是从你说的这边https://github.com/UKTailwind/PicoMiteAllVersions.git 拉的代码。

### 问题：
我刚刚编译了一下 Pico 1 的版本。会有存储 Map 定义溢出的问题。细看才发现 `configuration.h` 只有定义了 RP2350 的 配置。缺了 RP2040 的配置。两个版本可能要融合一下。

@2yanyi  如果纯纯想要 Pico2 跑 PicoMite 可以先checkout 我这个分支自己编。 @cuu 这个PR合到主分支还要放缓。尚需解决 memory map 定义问题。

adwuard, 2025-03-26:
> 我这个分支同步的代码，也是从你说的这边https://github.com/UKTailwind/PicoMiteAllVersions.git 拉的代码。
>
> ### 问题：
> 我刚刚编译了一下 Pico 1 的版本。会有存储 Map 定义溢出的问题。细看才发现 `configuration.h` 只有定义了 RP2350 的 配置。缺了 RP2040 的配置。两个版本可能要融合一下。 [...]

@cuu Pico 1 编译问题已解决，麻烦辛苦再合一下。https://github.com/cuu/PicoMite/pull/2

2yanyi, 2025-03-26:
> 我这个分支同步的代码，也是从你说的这边https://github.com/UKTailwind/PicoMiteAllVersions.git 拉的代码。
> ### 问题：
>
> 我刚刚编译了一下 Pico 1 的版本。会有存储 Map 定义溢出的问题。细看才发现 `configuration.h` 只有定义了 RP2350 的 配置。缺了 RP2040 的配置。两个版本可能要融合一下。 [...]

好的，感谢你的效率。不过使用了你的分支后我还是遇到了一些错误，例如在 PicoMite.c 中引用了 hardware/structs/ssi.h，但是 pico-sdk-2.1.1 中并没有这个文件。

```c
// hardware/structs/ssi.h
#ifndef rp2350
    #include "hardware/structs/ssi.h"
    #include "hardware/vreg.h"
#else
```

我还是等优化更新吧，辛苦！

altgreen64, 2025-04-01:
And what about these devices, picocal will be able to handle them?
https://milkv.io/duo
Luckfox Lyra ??

adwuard, 2025-04-01:
> And what about these devices, picocal will be able to handle them?
> https://milkv.io/duo
> Luckfox Lyra ??
> 
> śr., 26 mar 2025 o 15:17 言一。 ***@***.***> napisał(a):
> […](#)
@altgreen64 I do see people tryin to make Lyra work on the PicoCalc. Quite amazing.
https://forum.clockworkpi.com/t/luckfox-lyra-on-picocalc/16280/16