clockworkpi/PicoCalc #10: LVGL demo app issue with keyboard shift key and textarea

rhaamo, 2025-04-01:
I am working on a firmware that is based on the LVGL demo app in this repository and I have an issue with the keyboard.
I was stuck because when entering something like `1+1` I couldn´t get anything, same for `1*1`, but `-` or `/` would work...

Turns out the shift key does send something:
```
Key event a2
```

Which ends up in the textarea:
```
Key event a2
Key event 2b
Key event 31
Key event a
Textarea: '1�+1'
```

Is that an issue with the LVGL PicoCalc keyboard handling or expected ?

Extracts of my code where I use the textarea content:
```
// The textarea
    input = lv_textarea_create(screen);
    lv_obj_set_width(input, 320);
    lv_obj_set_height(input, LV_SIZE_CONTENT);    /// 20
    lv_obj_set_x(input, -10);
    lv_obj_set_y(input, 91);
    lv_obj_set_align(input, LV_ALIGN_CENTER);
    // Enable keyboard input for the text box
    lv_textarea_set_placeholder_text(input, "Input:");
    lv_textarea_set_one_line(input, true);
    lv_obj_set_style_anim_time(input, 5000, LV_PART_CURSOR|LV_STATE_FOCUSED);
    // Textarea event handler
    lv_obj_add_event_cb(input, textarea_event_handler, LV_EVENT_READY, input);

// The event handler
static void textarea_event_handler(lv_event_t *e) {
    printf("Textarea: '%s'", lv_textarea_get_text(ta));
}
```

adwuard, 2025-04-02:
@rhaamo hi, I’m the guy who implemented the lvgl demo. The keyboard is event is currently lousy implemented. Since there’s no good key code documentation on picocalc’s i2c keyboard. The key event may not be 1-1 behavior.
I could take a deeper look into kb source code and see if I can manage to reverse engineering the logic behind the keyboard keycode and fix your problem.

Edward

rhaamo, 2025-04-02:
I've looked a bit at the keyboard firmware for something else and yeah :D

For now I did a hackish workaround where I filter the entry to only keep ASCII chars in the textarea handler. it will work for the time being.

adwuard, 2025-04-02:
@rhaamo I've fixed the issue and created a more verbose key mapping with case switching. This should fix this issue.
This should merge to master shortly, or you could pull this PR commit #11  and test it out.

## Resolved
All special chars are now mapped properly.
[image]

adwuard, 2025-04-02:
The LVGL code is quite new. @rhaamo Thank you for trying it out and addressing the issue.👏

rhaamo, 2025-04-02:
Great thanks ! I will directly grab the files and test.

rhaamo, 2025-04-02:
Confirmed it works fine after using your file and removing my "workaround", thanks again!

PR #11 (2025-04-03): fix: LVGL kb port, keycode ASCII hex mapping issue
Fixes #10 issue

## Problem
Keyboard keycode read from I2C keyboard was lousy return to LVGL causing hex input issue. Described in the Issue ticket.
 
## Fixed
- More deterministic case match, case/switch out all the possible I2C keyboard events, mapping to LVGL as much as possible. and remaps others properly. 
- This verbose key case handling also enable user to handle special key event's themselves.
closed 2025-04-03 by cuu