clockworkpi/PicoCalc #12: fuzix image or image-building instructions for rp2350 (pico 2w) boards?

wez, 2025-04-05:
Could you share whatever patches or other things were necessary to get fuzix running on the picocalc?

The provided fuzix image does boot on the Pico H.

However, when I build stock fuzix for the pico 2 using `make TARGET=rpipico SUBTARGET=pico2 diskimage` in the fuzix repo , I can build it and boot it to the usb console, but it fails to find the sd card and halts before it activates the screen.

I'm assuming that there is some additional configuration required to make the kernel and/or filesystem work on the picocalc hardware!

wez, 2025-04-05:
and moments after I posted that, I found the patch in this repo:
https://github.com/clockworkpi/PicoCalc/blob/master/Code/FUZIX/fuzix.patch

I changed this hunk so that I can build for rp2350 without the cmake forcing it to rp2040:

```diff
diff --git a/Kernel/platform/platform-rpipico/CMakeLists.txt b/Kernel/platform/platform-rpipico/CMakeLists.txt
index 0081014c4..7a4e91d32 100644
--- a/Kernel/platform/platform-rpipico/CMakeLists.txt
+++ b/Kernel/platform/platform-rpipico/CMakeLists.txt
@@ -1,5 +1,12 @@
 cmake_minimum_required(VERSION 3.12)

+##PICO_BOARD pico PICO_PLATFORM=rp2040
+#set(PICO_BAORD pico)
+#set(PICO_PLATFORM rp2040)
+
+#set(PICO_BOARD pico2)
+#set(PICO_PLATFORM rp2350-arm-s)
+
```

wez, 2025-04-05:
I've automated building fuzix images for pico and pico2 over here: https://github.com/wez/FUZIX

cuu, 2025-04-06:
Sorry , I just see the issue now , in the past 12 hours ,I was out 

and now all included ,your pre-compiled uf2 and your updated code as patch ,fuzix1.1.patch

wez, 2025-04-07:
Thanks, I think we can close this now!

closed 2025-04-07 by wez