clockworkpi/PicoCalc #15: Question: Is there a launcher that would work on PicoCalc?

mrickey, 2025-04-08:
I looked at a few, but they say they are designed for a 16x2 display.

I would like the ability to load images or UF2s from the SD card so I can have different code available on my PicoCalc depending on my current project.

adwuard, 2025-04-09:
That's an interesting question. Currently, I don't see an example app for this. But I think it is definitely possible. 
Have a "bootloader like" mini program that load the firmware dynamically from SD to specific pico program entry address. Or update pico's program entry point. It's is definitely possible.

adwuard, 2025-04-09:
Great news, I implemented a Bootoader that successfully booted firmware from a SD card!! Now I would just need a directory UI for it. 

## loading firmware from SD
erasing pico flash and writting new firmware to an offset location
```c
size_t program_size = 0;
    uint8_t buffer[FLASH_SECTOR_SIZE] = {0};
    size_t len = 0;
    
    // Erase and program flash in FLASH_SECTOR_SIZE chunks
    while ((len = fread(buffer, 1, sizeof(buffer), fp)) > 0) {
        // Ensure we don't write beyond the application area
        if ((program_size + len) > MAX_APP_SIZE) {
            printf("Error: Attempted to write beyond application area\n");
            fclose(fp);
            return false;
        }
        
        uint32_t ints = save_and_disable_interrupts();
        flash_range_erase(SD_BOOT_FLASH_OFFSET + program_size, FLASH_SECTOR_SIZE);
        flash_range_program(SD_BOOT_FLASH_OFFSET + program_size, buffer, len);
        restore_interrupts(ints);

        program_size += len;
    }
    printf("Program loaded successfully!\n");
```

launching program from offset flash location.
```c
void __not_in_flash_func(launch_application_from)(uint32_t *app_location) {
    // https://vanhunteradams.com/Pico/Bootloader/Bootloader.html
    uint32_t *new_vector_table = app_location;
    volatile uint32_t *vtor = (uint32_t *)(PPB_BASE + M0PLUS_VTOR_OFFSET);
    *vtor = (uint32_t)new_vector_table;
    asm volatile (
        "msr msp, %0\n"
        "bx %1\n"
        :
        : "r" (new_vector_table[0]), "r" (new_vector_table[1])
        : );
}
```

One caveat is that......
This App will take up ~256K of flash space. And loading program's start address comes after that.
Which means, all program will have to be re-compiled with updated starting flash address.

adwuard, 2025-04-09:
New Project: https://github.com/adwuard/Picocalc_SD_Boot

I've gone ahead and build this SD Card Boot App. It comes with GUI interface and you can dynamically load firmware from the SD card. Note that there are still some memory mapping challenge I sill need to improve on. 

[image]