clockworkpi/PicoCalc #20: reading battery status via i2c always reports 0

wez, 2025-04-17:
I'm not sure if I'm doing something wrong here, but I can read keyboard data and change the backlight just fine, so my i2c stuff generally works.

I have this rust function to read the battery register; it sends `REG_ID_BAT` and then reads back 2 bytes:

```rust
const KBD_ADDR: u8 = 0x1f;
const REG_ID_BKL: u8 = 0x05;
const REG_ID_FIF: u8 = 0x09;
const REG_ID_BK2: u8 = 0x0a;
const REG_ID_BAT: u8 = 0x0b;
const REG_WRITE: u8 = 1u8 << 7;

async fn read_battery_pct(
    i2c_bus: &mut I2c<'_, I2C1, embassy_rp::i2c::Async>,
) -> Result<u8, embassy_rp::i2c::Error> {
    let mut buf = [0u8; 2];
    i2c_bus
        .write_read_async(KBD_ADDR, [REG_ID_BAT], &mut buf)
        .await?;

    Ok(buf[1])
}
```

The data I get back is always `[REG_ID_BAT, 0]`, which corresponds to:

https://github.com/clockworkpi/PicoCalc/blob/4c656946290b72432826a543fbf266cd6d4ee4c2/Code/picocalc_keyboard/picocalc_keyboard.ino#L140-L143

but with `current_bat_pcnt` set to 0.

That value always appears to be `0` for me; whether the batteries are installed, removed, charging or when the usb-c cable to the picocalc (not the pico) is attached to the computer or not.

Am I reading the schematic correctly? I think the STM MCU `Serial1` is connected to `UART1` (pins 8/9) on the pico, so I should be able to read the PMU debugging information by reading from that uart in my pico firmware?  I capture and log that data, but I haven't seen anything get printed there so far.

Should I be doing something differently to read the battery level?

wez, 2025-04-17:
Compiling and flashing the current version of the master branch fixed the reported battery level for me

wez, 2025-04-18:
I found that I also needed https://github.com/clockworkpi/PicoCalc/pull/22 to read the charging status, too

cuu, 2025-04-19:
#22  did not fix all

Right now I realized that current_bat_pcnt is not a good idea, when I try to fix in 65bf0c6 I totally forgot  the reg of REG_ID_BAT,
current battery percent can be stored in  reg of REG_ID_BAT

So I will remove current_bat_pcnt from code later and add bitSet(percent,7) to indicate charging status just like @wez  said

cuu, 2025-04-21:
a1847ee3381d238ff7319fb81badc3366999034d 

removed current_bat_pcnt now 

I've  updated picocalc_helloworld ,added new battery info code with charging status
in  3b84959ee650235730d10283f5a5589dfb55eb9a

I also find out that, when I press the battery into socket for the first time, there is a slight time that axp2101 chip will report battery percent to be zero ,that causes low_bat() to do PMU.shutdown();

So I add a low_bat_count to debounce to avoid

wez, 2025-04-21:
@cuu the code looks good!  Regarding https://github.com/clockworkpi/PicoCalc/pull/22, I think the issue with the report_bat function still exists. Should I rebase that PR and re-submit, or do you have other plans for this?

cuu, 2025-04-21:
@wez sorry I forgot the report_bat issue, now I commented out all report_bat calls, but allow me to leave the code 

295f7e7a8e437981806cc92a6f4ad0fbb312f6d3