clockworkpi/PicoCalc #24: feature request: expose power button and shutdown command via i2c

prokrypt, 2025-04-19:
Enables safe shutdown of system and paves the way for implementing sleep.

benklop, 2025-04-20:
I think a separate register is a better idea.

madcock, 2025-04-26:
Related:
https://forum.clockworkpi.com/t/plans-for-keyboard-battery-stm32-firmware/17009/8

prokrypt, 2025-04-26:
> A keycode would be nice: power button is a button and handling an event is handier than polling a register

I agree. Technically a separate register is the right way to do it as keyboard and power button (handled by the axp chip) are separate systems, but just yeeting off a keycode is both simple to implement and requires minimal changes to software (picomite, micropython, etc) to handle.

Of course, I say simple, but I probably won't be the one to implement it 🤪