clockworkpi/PicoCalc #36: README + USB C/serial port explanation is non-sensical

therealgglggl, 2025-07-18:
In the README it says:

"USB Type-C for power and multifunctional design, default is Pico’s serial port."

'default is Pico's serial port' ??? - this is grammatically wrong, was it meant to say 'and is the default serial port for PicoCalc' ?
The way it reads right now its implying the Pico's serial port is the default USB C connector which is non-sensical.

Its not the default serial for the Pico board (i.e. Pico H or W) - that would be their micro-USB connector.

Also, the USB C connector doesnt respond to connections during bootsel - only the microusb port on the Pico itself responds, so the USB port at least as shipped with factory defaults, only functions as a power connector.

cuu, 2025-07-19:
According to the schema, the TYPE-C port connected to the Pico's UART0 TX/RX with no DIP turned on

therealgglggl, 2025-07-19:
According to what schema and what dip switches? can you please be more
specific?